tscircuit/footprinter #46: Create better footprint errors

seveibar, 2024-10-01:
[image]

ChiragAgg5k, 2024-10-01:
Hey i would like to work on this issue, can u assign it to me?

seveibar, 2024-10-01:
@ChiragAgg5k go for it. Check out this vid that walks through the architecture if you feel confused https://blog.tscircuit.com/p/learn-how-to-contribute-to-tscircuits

ChiragAgg5k, 2024-10-01:
Thankss we look into it

ChiragAgg5k, 2024-10-01:
[image]

here is the improved error message. how does it look?

seveibar, 2024-10-02:
Not improved. What was the not found function?

ChiragAgg5k, 2024-10-02:
ohh okk.  i will add that

ChiragAgg5k, 2024-10-02:
[image]

seveibar, 2024-10-05:
Hey sorry for the delay @ChiragAgg5k ! The `Received "soup"` part is throwing me off. What if it was:

`Function not found for footprinter "invalid". Specify a valid function like .dip, .lr, .p etc.`

Also when you PR this please add a test that shows it works!!! ty!!

If you need a faster response, you can always ping me on the [discord](https://tscircuit/join) as well!!!

Ojas2095, 2026-10-31:
/attempt #46

PR opened: https://github.com/tscircuit/footprinter/pull/980

Implements the exact suggested error message format requested by @seveibar:
`Function not found for footprinter "invalid". Specify a valid function like .dip, .lr, .p etc.` across `.circuitJson()`, `.soup()`, and `.json()`, along with full regression test coverage in `tests/better-footprint-errors.test.ts`. 100% CI checks passing.