tscircuit/footprinter #64: stamp footprint

seveibar, 2024-10-15:
This is an example stamp footprint. This one has holes but they don't always have holes

[image]

seveibar, 2024-10-18:
From the pico 2 datasheet

[image]

seveibar, 2024-10-18:
note that the mounting holes (the large holes) are not part of the stamp

seveibar, 2024-10-18:
There is another type (it is different) for "receiving stamps", the "stamp" should be used when creating a stamp board like the pico 2, but if you're trying to place a stamp board onto your board you'll need a "stampreceiver" which looks like this

[image]

seveibar, 2024-10-18:
CC @anas-sarkez this is a pretty good one to get started, we'll use this for placing and creating module boards

anas-sarkez, 2024-10-20:
#68 
@seveibar should this pull request fixed this issue ? or we need to add anything more

seveibar, 2024-10-21:
@anas-sarkez mostly! We still need to implement `stampreceiver` which is a PCB where you put on the `stampboard`. It's a bit confusing but a keyboard will use a stampreceiver, and somebody designing a module would use the `stampboard`. They're slightly different footprints.

closed 2024-10-21 by anas-sarkez