tscircuit/footprinter #75: default stampboard and stampreceiver

ShiboSoftwareDev, 2024-11-08:
I want to be able to type stampboard/stampreceiver and see a default footprint without having to enter the number of pins

[image]

[image]

ShiboSoftwareDev, 2024-11-08:
@anas-sarkez 👀

anas-sarkez, 2024-11-09:
I agree with you some of footprints are have a constant number of pins which should not be configurable
that's is what I did with sot23-5 (sot235) which you can not pass how many pins you want
[image]

ShiboSoftwareDev, 2024-11-09:
yes @anas-sarkez this is how it looks:
[image]

seveibar, 2024-11-09:
Instead of having num_pins not exist as a parameter we could also use z.literal(5) etc. that way every footprint at least emits the number of pads

ShiboSoftwareDev, 2024-11-09:
@seveibar using the .json() method how can I differentiate between num_pins whether it's a literal type or not?

anas-sarkez, 2024-11-09:
[image]
I think here where .json() gets the params

ShiboSoftwareDev, 2024-11-09:
@seveibar oh I didn't understand that code until now, thanks. I really should put some time into reading the codebase and understanding it sometimes

seveibar, 2024-11-18:
/bounty $12

PR #83 (2024-11-20): Refactor all footprinter functions by updating its scheme, all footprints now have default value of num_pins
/claim #75
/closes #75
closed 2024-11-20 by anas-sarkez