tscircuit/footprinter #88: pushbutton default footprint don't work anymore; regression

ShiboSoftwareDev, 2024-12-03:
[image]

seveibar, 2024-12-05:
/bounty $10

AnasSarkiz, 2024-12-05:
@seveibar 
We have changed the pushbutton parameter zod type to `literal` in #83. You need to pass the `pushbutton` string

[image]

this is a test file

[image]

let me know if there are any changes needed
cc: @seveibar

seveibar, 2024-12-05:
I don't understand, shibo gave a valid footprinter string- what is the valid string if not that?

seveibar, 2024-12-05:
Oh are you saying that the parameters are just not customizable at all? @AnasSarkiz

seveibar, 2024-12-05:
@ShiboSoftwareDev can you detect if the zod schema contains a literal woth a default and just not set it in the footprinter string? Sorry for the confusion, i think i understand now (maybe)

AnasSarkiz, 2024-12-05:
yes, he can customize any of them. the only thing he can do is write  `pushbutton` as footprinter string