tscircuit/footprinter #93: On rotation of footprint in multiple of 45, the pads are missing.

imrishabh18, 2024-12-04:
[image]

```
export default () => {
  return (
    <board height="43.18mm" width="18.54mm">
      <chip
        manufacturerPartNumber="ATmega328P"
        name="U1"
        schX={0}
        schY={8}
        pcbX={0}
        pcbY={-8}
        footprint="qfp32"
        pcbRotation={45}
      />
    </board>
  )
}
```

imrishabh18, 2024-12-04:
/bounty $5

ShiboSoftwareDev, 2024-12-04:
[image]
actually only 90 degree rotation works

imrishabh18, 2024-12-04:
Yeah I know 90 multiples are working fine. But I was seeing few pcb boards and they were placing the chips in 45deg. So, probably we need to support that as well. I think probably it will more work to rotate the pads I am guessing

ShiboSoftwareDev, 2024-12-04:
@imrishabh18 can you delete this issue and open it in snippets? Abse found a fix but transferring issues doesn't transfer bounties so please reopen it there. fwiw I implemented rotated_rect in all relevant repos except the autorouter repo. Can you please investigate this issue as I couldn't find the problme? https://github.com/tscircuit/autorouting/issues/96

imrishabh18, 2024-12-04:
If not here then, I am guessing this logic needs to be implemented in `PcbViewer` about rotation of footprint. Why in snippets though? 🤔

cc: @seveibar

ShiboSoftwareDev, 2024-12-04:
@imrishabh18 the original problem is the pads not appearing which is fixed by updating pcb-viewer in snippets since I already implemented it in pcb-viewr, and he found another problem where the rotation direction is reversed which needs to be fixed in pcb-viewer. Sorry I didn't clarify earlier