tscircuit/footprinter #101: transistor footprint

ShiboSoftwareDev, 2024-12-28:


seveibar, 2024-12-28:
There are a couple variants of transistor footprints here, so before i bounty this someone should just add some details e.g. the string names that should be added

ShiboSoftwareDev, 2024-12-28:
@seveibar I just need an npn transistor nothing specific, also shouldn't the one who'd implement this worry about the details? 
[image]

seveibar, 2024-12-28:
There are a lot of possible transistor footprints, so i just want to scope appropriately. We can identify the 5 most common using jlcsearch

seveibar, 2024-12-28:
http://jlcsearch.tscircuit.com/components/list?subcategory_name=Transistors%20(NPN/PNP)

Lets say SOT-23 and SOT-457

[image]

seveibar, 2024-12-28:
It looks like we already have sot23, so we just need sot457 for this issue (but shibo you can ship with sot23)

seveibar, 2024-12-28:
/bounty $7

PR #258 (2025-05-26): add support sot457 footprint
fixes #101 
/claim #101