tscircuit/footprinter #164: Generated documentation

seveibar, 2025-02-08:
I need generated documentation for each type of footprint. It should be short but useful to AI e.g.

```md

## `dip`

### Syntax

`dip${pins}[_w${width}_h${height}_p${pitch}]`

### Examples

- `dip8_w2.54mm`

### Defaults

- `w`: `2.56mm` for less than 8 pins, `1.24mm` for more than 8 pins

### Notes

<!-- Anything inside property.describe("...") could go here -->

- p: pitch distance between centers of pins

```

CC @ShiboSoftwareDev @Abse2001 @AnasSarkiz @Anshgrover23 

We can use AI processors for the generated documentation as long as the model is lightweight, e.g. 4o-mini. If an AI processor is used, make sure to commit the cache so we're not re-making queries.

/bounty $40

AnasSarkiz, 2025-02-08:
/attempt #164

yashksaini-coder, 2025-02-28:
/attempt #164

ShiboSoftwareDev, 2025-02-28:
/attempt #164

Rishikesh63, 2025-03-01:
/attempt #164

seveibar, 2025-03-01:
CC @ShiboSoftwareDev @Rishikesh63 a couple people are attempting this and I'm not seeing anything usable. I don't want to have a high bounty on something that I don't think people are going to get something usable for so closing. I also have a lot of spend on openai today so this issue is more expensive than it's worth and bad docs are a major issue.

This is essentially the content we want:

https://docs.tscircuit.com/footprints/footprinter-strings

seveibar, 2025-03-01:
FWIW @ShiboSoftwareDev was definitely closest but I just don't see a path without significant effort and we don't have time for it

closed 2025-03-01 by seveibar