tscircuit/footprinter #180: Implement SPDIP-28

Abse2001, 2025-02-10:
Here's are components on JLCPCB with this footprint: https://jlcsearch.tscircuit.com/components/list?package=SPDIP-28

There may be reference dimensions/designs on the [kicad viewer](https://tscircuit.github.io/kicad-viewer)

This issue may be eligible for a $10 bounty from the TSCircuit team (please review!)

bitdeckai, 2026-02-23:
I can take this. Plan: add SPDIP-28 footprint parameters aligned to KiCad/JLC reference, wire it into the footprinter string mapping, and add a snapshot test to validate output.

bitdeckai, 2026-02-23:
Opened PR: https://github.com/tscircuit/footprinter/pull/508

selenaalpha77-sketch, 2026-05-01:
/attempt #180

Heyzerohey, 2026-05-24:
/claim #180 via https://github.com/tscircuit/footprinter/pull/643

Ahmed5754, 2026-09-29:
/attempt #180

Ahmed5754, 2026-09-29:
/claim #180 via https://github.com/tscircuit/footprinter/pull/899